roboticslab-uc3m/asibot-main #1: install openrave

edwinDOS, 2015-02-20:
Following the opnerave installation tutorial for ubuntu, when i type:
**sudo apt-get install libsoqt4-dev libcoin60-dev**
get the error:

*Leyendo lista de paquetes... Hecho*
*Creando árbol de dependencias*
*Leyendo la información de estado... Hecho*
*El paquete libcoin60-dev no está disponible, pero algún otro paquete hace referencia*
*a él. Esto puede significar que el paquete falta, está obsoleto o solo se*
*encuentra disponible desde alguna otro origen*

*E: El paquete «libcoin60-dev» no tiene un candidato para la instalación*

edwinDOS, 2015-02-20:
I installed Ubuntu 14.04.2 LTS

jgvictores, 2015-02-20:
It seems from http://packages.ubuntu.com/search?keywords=libcoin that libcoin60-dev was dropped from ubuntu 12.04 to 14.04. Try with **sudo apt-get install libsoqt4-dev libcoin80-dev** instead.

edwinDOS, 2015-02-20:
when i typed:
      **cd openrave; mkdir build; cd build; cmake ..; make -j3**
shows the next error:

```javascript
Linking CXX executable openrave0.9
/usr/bin/ld: CMakeFiles/openrave.dir/openrave.cpp.o: referencia sin definir al símbolo '_ZN5boost6system15system_categoryEv'
//usr/lib/x86_64-linux-gnu/libboost_system.so.1.54.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [src/openrave0.9] Error 1
make[1]: *** [src/CMakeFiles/openrave.dir/all] Error 2
make[1]: *** Se espera a que terminen otras tareas....
[ 97%] Building CXX object plugins/rplanners/CMakeFiles/rplanners.dir/parabolicsmoother.cpp.o
[ 98%] Building CXX object plugins/rplanners/CMakeFiles/rplanners.dir/pathoptimizers.cpp.o
[ 98%] Building CXX object plugins/rplanners/CMakeFiles/rplanners.dir/randomized-astar.cpp.o
[ 99%] Building CXX object plugins/rplanners/CMakeFiles/rplanners.dir/workspacetrajectorytracker.cpp.o
/home/dani/openrave/plugins/rplanners/workspacetrajectorytracker.cpp: In member function ‘virtual bool WorkspaceTrajectoryTracker::InitPlan(OpenRAVE::RobotBasePtr, OpenRAVE::PlannerBase::PlannerParametersConstPtr)’:
/home/dani/openrave/plugins/rplanners/workspacetrajectorytracker.cpp:74:27: warning: variable ‘tstate’ set but not used [-Wunused-but-set-variable]
                 Transform tstate = _manip->GetTransform();
                           ^
/home/dani/openrave/plugins/rplanners/workspacetrajectorytracker.cpp: In member function ‘virtual OpenRAVE::PlannerStatus WorkspaceTrajectoryTracker::PlanPath(OpenRAVE::TrajectoryBasePtr)’:
/home/dani/openrave/plugins/rplanners/workspacetrajectorytracker.cpp:163:31: warning: variable ‘fendtime’ set but not used [-Wunused-but-set-variable]
         dReal fstarttime = 0, fendtime = workspacetraj->GetDuration();
                               ^
Linking CXX shared library librplanners.so
[ 99%] Built target rplanners
make: *** [all] Error 2
```

jgvictores, 2015-02-21:
Okay, I've updated the libcoin version at 55f4b68d51687c5938ce318faf5adc3cbec605c1 and opened https://github.com/roboticslab-uc3m/asibot-main/issues/2 regarding a need to specify distro for install instructions.
Regarding your new issue, try switching to the openrave master branch, maybe it has been fixed:
**git checkout master; git pull**

jgvictores, 2015-02-26:
Was solved by installing all boost: **sudo apt-get install libboost-all-dev**, and then performing clean OpenRAVE build: **cd openrave; rm -rf build; mkdir build; cmake ..; make -j3**

closed 2015-02-26 by jgvictores