roboticslab-uc3m/asibot-main #2: specify distro version in installation guides

jgvictores, 2015-02-21:
Similar has been done at https://github.com/roboticslab-uc3m/teo-main/tree/master/doc

PeterBowman, 2016-10-21:
@jgvictores is Markdown format preferred over Doxygen (see [teo-main/doc](https://github.com/roboticslab-uc3m/teo-main/tree/830d497eec5e1cc72a5833be74cdd4cd7167fe97/doc) and its [latest commits](https://github.com/roboticslab-uc3m/teo-main/commits/master/doc))? Should these files be accessed from GitHub or robots.uc3m.es/dox-...?

jgvictores, 2016-10-22:
>  is Markdown format preferred over Doxygen (see teo-main/doc and its latest commits)?

Yes, Markdown is preferred over Doxygen for install guides, and essentially anything that is not pure class/library/program documentation.

> Should these files be accessed from GitHub or robots.uc3m.es/dox-...?

Precisely, the intention is to **not** have to rely on CRON of robots.uc3m.es and instead get immediate GitHub visualization. Take a look at the raw of the [teo-main README](https://github.com/roboticslab-uc3m/teo-main/blob/464b8cd0f70c019feacf3b9124f6cbcbb460cc13/README.md):
```
Installation instructions for installing from source can be found [here](doc/teo_install_on_ubuntu_14_10.md).
```

jgvictores, 2016-10-27:
@PeterBowman Anything needed here?
It is set as a requirement for #23

jgvictores, 2016-10-27:
@PeterBowman BTW, I started https://github.com/roboticslab-uc3m/asibot-main/projects/1 because it really helps to keep track of priorities and progress.

PeterBowman, 2016-10-28:
@jgvictores I took advantage of the fact that all three issues are related and some work was already done in a single branch. A PR seemed more suited for keeping track and requesting feedback. Maybe it wasn't the best choice, though.

jgvictores, 2016-10-28:
No problem, was just giving info.
Anyway, can we close this issue given the PR?

PeterBowman, 2016-10-28:
Sure, if you think that's more appropriate to do in this case. Ideally, the issue should auto-close thanks to [this feature](https://help.github.com/articles/closing-issues-via-commit-messages/) when the relevant commit is merged (haven't tested it yet). The intention was to leave all three issues open to collect general input, and gather any implementation-focused comments in that single PR. Then, once everything is ready, a simple merge to develop should do the trick. Sorry for not clarifying that earlier.

jgvictores, 2016-10-28:
Mmm I'll then interpret this is still open,  as the close commit hasn't arrived. In consequence, #23 should not be merged yet. Please correct me if my interpretation is wrong! :smiley:

PeterBowman, 2016-10-28:
That's correct, in the next days I'll submit a patch to #23 that will (hopefully) resolve this issue.

commit ac882357 (2016-11-02): Migrate installation guides to Markdown format

Follows up 9c62679, resolves #2. This change set was split into two
separate commits to facilitate proper diff visualization on GitHub (file
moves with significant changes were shown as add+delete instead of plain
move).
closed 2016-11-02 by jgvictores