roboticslab-uc3m/asibot-main #3: cartesianServer doesn't work properly

iagson, 2015-02-23:
Hello jcgvitores,

I was trying to start an instance of a cartesianServer, to see if all software was installed correctly and the kitchen  appears to initialise upside down. That isn't so bad, but when you try to input some extra commands like " cartesianServer --env asibot_kitchen_cameras.env.xml " the GUI doesn't start beacuse of an error (already tried by changing the .ini file)

These are the screen prints I took:
[image]
[image]

Thank you very much.

jgvictores, 2015-02-23:
1) On your first screenshot I see a YARP matching protocol error (1 0 0 vs 0 0 0). You should be able to fix that by cleaning and recompiling (may be overkill, but I'd first do it on YARP and then on ASIBOT).
2) Check if the environment that breaks has cameras. There is a known issue regarding the off-screen rendering we use and certain graphic card drivers (especially in Linux).

BTW, don't mind copying and pasting the entire log, it will make things easier!

iagson, 2015-02-23:
After recompiling both YARP and ASIBOT the error persists.
BTW, I had to rename the asibot-main directory to asibot so to initialise cartesianServer.
This is the whole log after trying to access a camera:

iagson@iagson-VirtualBox:~$ cartesianServer --env asibot_kitchen_cameras.env.xml
||| clearing context
||| adding context [cartesianServer/conf]
||| configuring
||| policy set to ASIBOT_ROOT
||| ASIBOT_ROOT: /home/iagson/asibot
||| loading policy from /home/iagson/asibot/ASIBOT_ROOT.ini
yarp: cannot read from /home/iagson/asibot/ASIBOT_ROOT.ini
||| loading policy from /home/iagson/.yarp/ASIBOT_ROOT.ini
yarp: cannot read from /home/iagson/.yarp/ASIBOT_ROOT.ini
||| loading policy from /etc/yarp/policies/ASIBOT_ROOT.ini
yarp: cannot read from /etc/yarp/policies/ASIBOT_ROOT.ini
||| loading policy from /etc/ASIBOT_ROOT.ini
yarp: cannot read from /etc/ASIBOT_ROOT.ini
||| loading policy from /usr/local/etc/ASIBOT_ROOT.ini
yarp: cannot read from /usr/local/etc/ASIBOT_ROOT.ini
||| failed to load policy from /home/iagson/asibot/ASIBOT_ROOT.ini /home/iagson/.yarp/ASIBOT_ROOT.ini /etc/yarp/policies/ASIBOT_ROOT.ini /etc/ASIBOT_ROOT.ini /usr/local/etc/ASIBOT_ROOT.ini
||| default config file specified as cartesianServer.ini
||| checking [/home/iagson/cartesianServer.ini] (pwd)
||| checking [/home/iagson/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/iagson/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/iagson/asibot/app/cartesianServer/conf/cartesianServer.ini] (deprecated-old-style-context)
||| found /home/iagson/asibot/app/cartesianServer/conf/cartesianServer.ini
Run "cartesianServer --help" for options
cartesianServer checking for yarp network... [ok]
--------------------------------------------------------------
CartesianServer using controller: cartesianbot,  prefix: /ravebot.
CartesianServer using movjLocal: /ravebot/movjCartesianServer, movjRemote: /ravebot.
--------------------------------------------------------------
CartesianBot using A0: 0.300000, A1: 0.400000, A2: 0.400000, A3: 0.300000.
CartesianBot using cmcMs: 30.000000, duration: 20.000000, maxAcc: 0.200000, maxVel: 7.500000.
CartesianBot using robotDevice: controlboard, robotSubDevice: ravebot, robotName: /ravebot.
CartesianBot using robotLocal: N/A, robotRemote: N/A.
[INFO]ControlBoard subdevice is ravebot
--------------------------------------------------------------
RaveBot using numMotors: 5, env: asibot_kitchen_cameras.env.xml, externObj: none, extraRobot: none.
RaveBot using genInitPos: 0.000000, genJointTol: 0.250000, genMaxLimit: 360.000000, genMinLimit: -360.000000.
RaveBot using genRefSpeed: 7.500000, genEncRawExposed: 0.017453, genVelRawExposed: 0.017453.
RaveBot using jmcMs: 20.000000, jmcMsAcc: 1.000000, modePosVel: 0, physics: none, viewer: 1.
RaveBot not using individual initPoss, defaulting to genInitPos.
RaveBot not using individual jointTols, defaulting to genJointTol.
RaveBot not using individual maxLimits, defaulting to genMaxLimit.
RaveBot not using individual minLimits, defaulting to genMinLimit.
RaveBot not using individual refSpeeds, defaulting to genRefSpeed.
RaveBot not using individual encRawExposeds, defaulting to genEncRawExposed.
RaveBot not using individual velRawExposeds, defaulting to genVelRawExposed.
--------------------------------------------------------------
[plugindatabase.h:577 Create] Failed to create name viewerrecorder, interface module
[environment-core.h:800 SetPhysicsEngine] setting ode physics engine
[odephysics.h:255 InitEnvironment] ode params: erp=1.000000e-02 (2.000000e-01), cfm=1.000000e-05 (1.000000e-10), itrs=20 (20)
[success] RaveBot loaded environment.
RaveBot using robot 0 (asibot) as main robot.
Sensors found on robot 0 (asibot): 1.
Sensor 0 name: asibot_tip
Sensor 0 supports ST_Camera.
[... 33 lines cut ...]
received an unknown request after a VOCAB_GET: prot
ERROR, expecting protocol 1 0 0, but remotecontrolboard returned protocol version 0 0 0
WARNING: ignoring error but please update YARP or the remotecontrolboard implementation
[RaveBot] getAxes reporting 5 axes are present
[INFO]created device <remote_controlboard>. See C++ class yarp::dev::RemoteControlBoard for documentation.
[CartesianServer] success: Acquired robot interfaces.
yarp: Port /ravebot/cartesianServer/rpc:i active at tcp://10.0.2.15:10010
yarp: Port /ravebot/cartesianServer/command:i active at tcp://10.0.2.15:10011
Coin info in glglue_allow_newer_opengl(): 

Features of OpenGL version > 1.0 has been
disabled, due to the use of a remote display.

This is so because many common OpenGL drivers
have problems in this regard.

To force full OpenGL use, set the environment
variable COIN_FULL_INDIRECT_RENDERING=1 and re-run the application.

If you don't want this message displayed again,
set the environment variable COIN_DONT_INFORM_INDIRECT_RENDERING=1.

[qtcoinviewer.cpp:3417 _GetCameraImage] offscreen renderer failed (check video driver), disabling
[qtcoinviewer.cpp:3370 _GetCameraImage] cannot render offscreen
Segmentation fault (core dumped)

jgvictores, 2015-02-23:
The final segmentation fault is due to "offscreen renderer failed (check video driver)".
Here's a related issue: http://openrave-users-list.185357.n3.nabble.com/qtcoin-viewer-error-quot-offscreen-renderer-failed-quot-td4025805.html
This is an old issue related to OpenRAVE and deeper. At the asibot-main level, all you can do is avoid simulated environments with cameras.

jgvictores, 2017-12-05:
Throughout old emails, I've commented that offscreen rendering was achived via Bumblebee's `optirun` (only applicable to Optimus NVIDIA technology I guess).

jgvictores, 2017-12-05:
This issue was moved to roboticslab-uc3m/openrave-yarp-plugins#48

closed 2017-12-05 by jgvictores