roboticslab-uc3m/asibot-main #4: OpenRAVE seems to fail when initiating cartesianServer

iagson, 2015-02-27:
Hi jgvictores

I encountered another issue while running the cartesianServer instance. I did a fresh install of Ubuntu 14.10 on my computer (I previously had it on a Virtual Machine) and now I can't run the instance. OpenRAVE is the one that seems to fail, but I reinstalled it a couple times (without uninstalling first) even with the cmake videorecording-option Off.

I leave here the message I get when running cartesianServer:

iagson@iagsons-MagicBox:~$ cartesianServer
||| clearing context
||| adding context [cartesianServer/conf]
||| configuring
||| policy set to ASIBOT_ROOT
||| ASIBOT_ROOT: /home/iagson/asibot-main
||| loading policy from /home/iagson/asibot-main/ASIBOT_ROOT.ini
yarp: cannot read from /home/iagson/asibot-main/ASIBOT_ROOT.ini
||| loading policy from /home/iagson/.yarp/ASIBOT_ROOT.ini
yarp: cannot read from /home/iagson/.yarp/ASIBOT_ROOT.ini
||| loading policy from /etc/yarp/policies/ASIBOT_ROOT.ini
yarp: cannot read from /etc/yarp/policies/ASIBOT_ROOT.ini
||| loading policy from /etc/ASIBOT_ROOT.ini
yarp: cannot read from /etc/ASIBOT_ROOT.ini
||| loading policy from /usr/local/etc/ASIBOT_ROOT.ini
yarp: cannot read from /usr/local/etc/ASIBOT_ROOT.ini
||| failed to load policy from /home/iagson/asibot-main/ASIBOT_ROOT.ini /home/iagson/.yarp/ASIBOT_ROOT.ini /etc/yarp/policies/ASIBOT_ROOT.ini /etc/ASIBOT_ROOT.ini /usr/local/etc/ASIBOT_ROOT.ini
||| default config file specified as cartesianServer.ini
||| checking [/home/iagson/cartesianServer.ini] (pwd)
||| checking [/home/iagson/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/iagson/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/iagson/asibot-main/app/cartesianServer/conf/cartesianServer.ini] (deprecated-old-style-context)
||| found /home/iagson/asibot-main/app/cartesianServer/conf/cartesianServer.ini
Run "cartesianServer --help" for options.
cartesianServer checking for yarp network... [ok]
--------------------------------------------------------------
CartesianServer using controller: cartesianbot,  prefix: /ravebot.
CartesianServer using movjLocal: /ravebot/movjCartesianServer, movjRemote: /ravebot.
--------------------------------------------------------------
CartesianBot using A0: 0.300000, A1: 0.400000, A2: 0.400000, A3: 0.300000.
CartesianBot using cmcMs: 30.000000, duration: 20.000000, maxAcc: 0.200000, maxVel: 7.500000.
CartesianBot using robotDevice: controlboard, robotSubDevice: ravebot, robotName: /ravebot.
CartesianBot using robotLocal: N/A, robotRemote: N/A.
[INFO]ControlBoard subdevice is ravebot
--------------------------------------------------------------
RaveBot using numMotors: 5, env: asibot_kitchen.env.xml, externObj: none, extraRobot: none.
RaveBot using genInitPos: 0.000000, genJointTol: 0.250000, genMaxLimit: 360.000000, genMinLimit: -360.000000.
RaveBot using genRefSpeed: 7.500000, genEncRawExposed: 0.017453, genVelRawExposed: 0.017453.
RaveBot using jmcMs: 20.000000, jmcMsAcc: 1.000000, modePosVel: 0, physics: none, viewer: 1.
RaveBot not using individual initPoss, defaulting to genInitPos.
RaveBot not using individual jointTols, defaulting to genJointTol.
RaveBot not using individual maxLimits, defaulting to genMaxLimit.
RaveBot not using individual minLimits, defaulting to genMinLimit.
RaveBot not using individual refSpeeds, defaulting to genRefSpeed.
RaveBot not using individual encRawExposeds, defaulting to genEncRawExposed.
RaveBot not using individual velRawExposeds, defaulting to genVelRawExposed.
--------------------------------------------------------------
cartesianServer: symbol lookup error: cartesianServer: undefined symbol: _ZN8OpenRAVE21RaveCreateEnvironmentEv
iagson@iagsons-MagicBox:~$ echo "export ASIBOT_ROOT=$PWD/asibot-main" >> ~/.profile
iagson@iagsons-MagicBox:~$ echo "export ASIBOT_DIR=$PWD/asibot-main/build" >> ~/.profile
iagson@iagsons-MagicBox:~$ source ~/.profile
iagson@iagsons-MagicBox:~$ cartesianServer
||| clearing context
[... 39 lines cut ...]
CartesianBot using cmcMs: 30.000000, duration: 20.000000, maxAcc: 0.200000, maxVel: 7.500000.
CartesianBot using robotDevice: controlboard, robotSubDevice: ravebot, robotName: /ravebot.
CartesianBot using robotLocal: N/A, robotRemote: N/A.
[INFO]ControlBoard subdevice is ravebot
--------------------------------------------------------------
RaveBot using numMotors: 5, env: asibot_kitchen.env.xml, externObj: none, extraRobot: none.
RaveBot using genInitPos: 0.000000, genJointTol: 0.250000, genMaxLimit: 360.000000, genMinLimit: -360.000000.
RaveBot using genRefSpeed: 7.500000, genEncRawExposed: 0.017453, genVelRawExposed: 0.017453.
RaveBot using jmcMs: 20.000000, jmcMsAcc: 1.000000, modePosVel: 0, physics: none, viewer: 1.
RaveBot not using individual initPoss, defaulting to genInitPos.
RaveBot not using individual jointTols, defaulting to genJointTol.
RaveBot not using individual maxLimits, defaulting to genMaxLimit.
RaveBot not using individual minLimits, defaulting to genMinLimit.
RaveBot not using individual refSpeeds, defaulting to genRefSpeed.
RaveBot not using individual encRawExposeds, defaulting to genEncRawExposed.
RaveBot not using individual velRawExposeds, defaulting to genVelRawExposed.
--------------------------------------------------------------
cartesianServer: symbol lookup error: cartesianServer: undefined symbol: _ZN8OpenRAVE21RaveCreateEnvironmentEv

AND HERE I leave the execution of openrave:

iagson@iagsons-MagicBox:~$ openrave
[textserver.h:403 main] text server listening on port 4765
[plugindatabase.h:577 Create] Failed to create name qtcoin, interface viewer
[openrave.cpp:315 MainOpenRAVEThread] failed to find an OpenRAVE viewer.

jgvictores, 2015-03-04:
I'd try to fix:
`[ERROR]expecting protocol 1 1 0, but remotecontrolboard returned protocol version 0 0 0`
This can happen if different parts of a program are compiled with different versions of YARP. Try the following:
* Make sure you only have one installation of YARP.
* Completely uninstall and delete any trace of asibot-main.
* Perform a fresh clean install of asibot-main against the unique YARP installation.

iagson, 2015-03-10:
Hello jgvictores,

I tried what you suggested with no luck, even formatting the OS, but the cartesianServer keeps failing.
This time the error message that you prompted last time has changed to:

*[ERROR]expecting protocol 1 2 0, but remotecontrolboard returned protocol version 0 0 0*

and here is the rest of the log, although it's very similar to the previous one:

iagson@iagson-ControlMachine:~$ cartesianServer
||| clearing context
||| adding context [cartesianServer/conf]
||| configuring
||| policy set to ASIBOT_ROOT
||| ASIBOT_ROOT: /home/iagson/asibot
||| loading policy from /home/iagson/asibot/ASIBOT_ROOT.ini
yarp: cannot read from /home/iagson/asibot/ASIBOT_ROOT.ini
||| loading policy from /home/iagson/.yarp/ASIBOT_ROOT.ini
yarp: cannot read from /home/iagson/.yarp/ASIBOT_ROOT.ini
||| loading policy from /etc/yarp/policies/ASIBOT_ROOT.ini
yarp: cannot read from /etc/yarp/policies/ASIBOT_ROOT.ini
||| loading policy from /etc/ASIBOT_ROOT.ini
yarp: cannot read from /etc/ASIBOT_ROOT.ini
||| loading policy from /usr/local/etc/ASIBOT_ROOT.ini
yarp: cannot read from /usr/local/etc/ASIBOT_ROOT.ini
||| failed to load policy from /home/iagson/asibot/ASIBOT_ROOT.ini /home/iagson/.yarp/ASIBOT_ROOT.ini /etc/yarp/policies/ASIBOT_ROOT.ini /etc/ASIBOT_ROOT.ini /usr/local/etc/ASIBOT_ROOT.ini
||| default config file specified as cartesianServer.ini
||| checking [/home/iagson/cartesianServer.ini] (pwd)
||| checking [/home/iagson/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/iagson/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/iagson/asibot/app/cartesianServer/conf/cartesianServer.ini] (deprecated-old-style-context)
||| found /home/iagson/asibot/app/cartesianServer/conf/cartesianServer.ini
Run "cartesianServer --help" for options.
cartesianServer checking for yarp network... [ok]
--------------------------------------------------------------
CartesianServer using controller: cartesianbot,  prefix: /ravebot.
CartesianServer using movjLocal: /ravebot/movjCartesianServer, movjRemote: /ravebot.
--------------------------------------------------------------
CartesianBot using A0: 0.300000, A1: 0.400000, A2: 0.400000, A3: 0.300000.
CartesianBot using cmcMs: 30.000000, duration: 20.000000, maxAcc: 0.200000, maxVel: 7.500000.
CartesianBot using robotDevice: controlboard, robotSubDevice: ravebot, robotName: /ravebot.
CartesianBot using robotLocal: N/A, robotRemote: N/A.
[INFO]ControlBoard subdevice is ravebot
--------------------------------------------------------------
RaveBot using numMotors: 5, env: asibot_kitchen.env.xml, externObj: none, extraRobot: none.
RaveBot using genInitPos: 0.000000, genJointTol: 0.250000, genMaxLimit: 360.000000, genMinLimit: -360.000000.
RaveBot using genRefSpeed: 7.500000, genEncRawExposed: 0.017453, genVelRawExposed: 0.017453.
RaveBot using jmcMs: 20.000000, jmcMsAcc: 1.000000, modePosVel: 0, physics: none, viewer: 1.
RaveBot not using individual initPoss, defaulting to genInitPos.
RaveBot not using individual jointTols, defaulting to genJointTol.
RaveBot not using individual maxLimits, defaulting to genMaxLimit.
RaveBot not using individual minLimits, defaulting to genMinLimit.
RaveBot not using individual refSpeeds, defaulting to genRefSpeed.
RaveBot not using individual encRawExposeds, defaulting to genEncRawExposed.
RaveBot not using individual velRawExposeds, defaulting to genVelRawExposed.
--------------------------------------------------------------
[plugindatabase.h:577 Create] Failed to create name viewerrecorder, interface module
[environment-core.h:800 SetPhysicsEngine] setting ode physics engine
[odephysics.h:255 InitEnvironment] ode params: erp=1.000000e-02 (2.000000e-01), cfm=1.000000e-05 (1.000000e-10), itrs=20 (20)
[... 33 lines cut ...]
checkProtocolVersion failed
yarp: Removing output from /ravebot/movjCartesianServer/rpc:o to /ravebot/rpc:i
yarp: Removing input from /ravebot/movjCartesianServer/rpc:o to /ravebot/rpc:i
yarp: Removing input from /ravebot/movjCartesianServer/command:o to /ravebot/command:i
yarp: Removing output from /ravebot/movjCartesianServer/command:o to /ravebot/command:i
yarp: Removing input from /ravebot/state:o to /ravebot/movjCartesianServer/state:i
yarp: Removing output from /ravebot/state:o to /ravebot/movjCartesianServer/state:i
yarpdev: ***ERROR*** driver <remote_controlboard> was found but could not open
[error] Class instantiation not worked.

[error] movjRemote not valid, has a remotely accessible device been open?

Module failed to open
[INFO]Server control board stopping
RaveBot::close() Closing ports...
[WorldRpcResponder] Got 
[WorldRpcResponder] Got 
[qtcoinviewer.cpp:3041 UpdateFromModel] auto-setting camera location
RaveBot::close() Ports closed. Closing environment...
[environment-core.h:221 Destroy] destroy module
[environment-core.h:237 Destroy] quitting viewer qtcoin
[qtcoinviewer.cpp:375 ~QtCoinViewer] destroying qtcoinviewer
RaveBot: close() joining...
Segmentation fault (core dumped)
iagson@iagson-ControlMachine:~$

jgvictores, 2015-03-11:
Should be fixed at fc26336c24434f6c0ecbdd283ec1ebb9b51793f8, will now clean up and merge with master.

closed 2015-03-11 by jgvictores