roboticslab-uc3m/asibot-main #7: [environment-core.h:475 Load] load failed on file  [error] RaveBot could not load  environment.

serna92, 2016-04-01:
Hello jgvictores,

I'm using Ubuntu 14.04 and I compiled  the asibot-main repository and executed "cartesian server" command after installing it by the following tutorial:

http://robots.uc3m.es/dox-asibot-main/install_on_ubuntu.html

I attach the output of the command "cmake" and the command "cartesian server".

I also have problems with Intel drivers but it seems that there is no support for Ubuntu Trusty.

[cmake.txt](https://github.com/roboticslab-uc3m/asibot-main/files/199713/cmake.txt)
[cartesian_server.txt](https://github.com/roboticslab-uc3m/asibot-main/files/199714/cartesian_server.txt)

jgvictores, 2016-04-08:
Could you copy and paste the output of:
`ls /usr/local/share/asibot/contexts/ravebot/models/`
?

Just for the record, no need to touch anything yet, here are some pointers for your additional issues:
* WARNING: your [cmake.txt](https://github.com/roboticslab-uc3m/asibot-main/files/199713/cmake.txt) Makes me believe you may have two installations of Boost. The one in /usr/local usually means you manually installed it [(read more, in spanish)](http://asrob.uc3m.es/index.php/Estructura_de_carpetas_en_Debian_y_derivados_%28Ubuntu,_Mint...%29). If so, the best way would be to perform an uninstall using the files you used for installation (`sudo make uninstall` from the build directory, but I'm not too sure if Boost still uses its bjam system). Then rm asibot-main/build and try again.
* WARNING: your [cartesian_server.txt](https://github.com/roboticslab-uc3m/asibot-main/files/199714/cartesian_server.txt) indicates something wrong in the OSG->openrave->asibot-main chain. I'd uninstall asibot-main and openrave (in that order), then fresh compile and install openrave and asibot-main (in that order). `sudo ldconfig` can also help, to keep the dynamic library [(read more, in spanish)](http://asrob.uc3m.es/index.php/Librer%C3%ADas_Est%C3%A1ticas_y_Din%C3%A1micas) indexing updated.

serna92, 2016-04-08:
That's the output of ls `/usr/local/share/asibot/contexts/ravebot/models/`

asibot.kinbody.xml                 ipcamCart.kinbody.xml
asibot_kitchen_cameras2.env.xml    ipcamCart.robot.xml
asibot_kitchen_cameras.env.xml     ipcam.kinbody.xml
asibot_kitchen.env.xml             ipcam.robot.xml
asibot_kitchen_force6d.env.xml     ivs
asibot_kitchen_gafasAR.env.xml     kinect.kinbody.xml
asibot_kitchen_head.env.xml        kinect.robot.xml
asibot_kitchen_kinect.env.xml      kitchen.kinbody.xml
asibot_kitchen_planning.env.xml    redCan.kinbody.xml
asibot_kitchen_robotDraw.env.xml   VS2.env.xml
asibot_kitchen_v2.env.xml          VS3.env.xml
asibot_kitchen_VS.env.xml          VS4.env.xml
asibot_kitchen_wheelchair.env.xml  VS.env.xml
asibotLW.kinbody.xml               wheelchair_gafasAR.kinbody.xml
asibotLW.robot.xml                 wheelchair_gafasAR.robot.xml
asibot.robot.xml                   wheelchair.kinbody.xml
docking.kinbody.xml                wheelchairLW.kinbody.xml
felix2.env.xml                     wheelchairLW.robot.xml
felix.env.xml                      wheelchairR.kinbody.xml
ipcamCart2.kinbody.xml             wheelchair.robot.xml
ipcamCart2.robot.xml               wheelchairR.robot.xml
ipcamCart3.kinbody.xml             wrls
ipcamCart3.robot.xml

I will check the additional issues right now.

jgvictores, 2016-04-08:
The other ones are slower, this should be possible without them. Please copy+paste results of:
`ls /usr/local/share/yarp/config/path.d`
`cat /usr/local/share/yarp/config/path.d/*`
`echo $YARP_DATA_DIRS`

serna92, 2016-04-08:
`ls /usr/local/share/yarp/config/path.d`
`cat /usr/local/share/yarp/config/path.d/*`

ls: no se puede acceder a /usr/local/share/yarp/config/path.d: No existe el archivo o el directorio

`echo $YARP_DATA_DIRS`

It doesn't appear nothing, maybe I haven't it setted

jgvictores, 2016-04-08:
Can you `cd` to `asibot-main/build`, and copy and paste here the output of `sudo make install`?

serna92, 2016-04-08:
[  3%] Built target TinyMath
[  7%] Built target TestRaveBotLib
[ 10%] Built target CartesianClientLib
[ 21%] Built target CartesianServerLib
[ 25%] Built target ColorClientLib
[ 64%] Built target ravebot
[ 82%] Built target cartesianbot
[ 85%] Built target RlPlugins
[ 96%] Built target TaskLib
[100%] Built target cartesianServer
Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/local/share/asibot/ymanager.ini
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/wheelchair_gafasAR.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_kinect.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/kinect.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/wheelchairLW.robot.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_force6d.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/kinect.robot.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/VS3.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibotLW.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ipcam.robot.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibotLW.robot.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ipcamCart.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ipcamCart2.robot.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_head.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/kitchen.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot.robot.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/wheelchairR.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_gafasAR.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_v2.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ipcam.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/felix2.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_robotDraw.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_wheelchair.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_VS.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ipcamCart2.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/VS4.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_planning.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/VS.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/redCan.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/VS2.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/wheelchair.robot.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/felix.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ipcamCart.robot.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/wheelchairLW.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/wheelchairR.robot.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_cameras2.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ipcamCart3.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/docking.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_cameras.env.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/wheelchair.kinbody.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/wheelchair_gafasAR.robot.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ipcamCart3.robot.xml
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ivs/ensam01sn.iv
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ivs/machosn.iv
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ivs/ensam02sn.iv
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ivs/dstation.iv
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ivs/ensam03sn.iv
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ivs/wheelchair.iv
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ivs/cocina.iv
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ivs/elmacho.iv
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ivs/ensam04sn.iv
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ivs/docking.iv
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/ivs/lata.iv
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/wrls/wheelchairRheadless.wrl
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/wrls/kinect.wrl
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/wrls/wheelchairElder.wrl
-- Up-to-date: /usr/local/share/asibot/contexts/ravebot/models/wrls/wheelchairRhead.wrl
-- Up-to-date: /usr/local/share/asibot/contexts/cartesianServer/cartesianServer.ini
[... cut ...]
-- Up-to-date: /usr/local/share/asibot/contexts/cartesianServer/cartesianServerKdl.ini
-- Up-to-date: /usr/local/share/asibot/contexts/cartesianServer/cartesianServerPXA270.ini
-- Up-to-date: /usr/local/lib/libTestRaveBotLib.a
-- Up-to-date: /usr/local/lib/libCartesianClientLib.a
-- Up-to-date: /usr/local/lib/libCartesianServerLib.a
-- Up-to-date: /usr/local/lib/libColorClientLib.a
-- Up-to-date: /usr/local/lib/libRlPlugins.a
-- Up-to-date: /usr/local/lib/libravebot.a
-- Up-to-date: /usr/local/lib/libcartesianbot.a
-- Up-to-date: /usr/local/bin/cartesianServer

jgvictores, 2016-04-08:
Ok, normally there should be a:
`-- Up-to-date: /usr/local/share/yarp/config/path.d/asibot.ini`
before your:
`-- Up-to-date: /usr/local/share/asibot/ymanager.ini`

Please uninstall asibot-main and yarp, then fresh reinstall yarp and asibot-main.

serna92, 2016-04-08:
`sudo make install`

[  3%] Built target TinyMath
[  7%] Built target TestRaveBotLib
[ 10%] Built target CartesianClientLib
[ 21%] Built target CartesianServerLib
[ 25%] Built target ColorClientLib
[ 64%] Built target ravebot
[ 82%] Built target cartesianbot
[ 85%] Built target RlPlugins
[ 96%] Built target TaskLib
[100%] Built target cartesianServer
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/share/asibot/ymanager.ini
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/wheelchair_gafasAR.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_kinect.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/kinect.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/wheelchairLW.robot.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_force6d.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/kinect.robot.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/VS3.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibotLW.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ipcam.robot.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibotLW.robot.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ipcamCart.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ipcamCart2.robot.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_head.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/kitchen.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot.robot.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/wheelchairR.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_gafasAR.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_v2.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ipcam.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/felix2.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_robotDraw.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_wheelchair.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_VS.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ipcamCart2.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/VS4.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_planning.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/VS.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/redCan.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/VS2.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/wheelchair.robot.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/felix.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ipcamCart.robot.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/wheelchairLW.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/wheelchairR.robot.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_cameras2.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ipcamCart3.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/docking.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/asibot_kitchen_cameras.env.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/wheelchair.kinbody.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/wheelchair_gafasAR.robot.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ipcamCart3.robot.xml
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ivs/ensam01sn.iv
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ivs/machosn.iv
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ivs/ensam02sn.iv
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ivs/dstation.iv
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ivs/ensam03sn.iv
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ivs/wheelchair.iv
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ivs/cocina.iv
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ivs/elmacho.iv
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ivs/ensam04sn.iv
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ivs/docking.iv
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/ivs/lata.iv
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/wrls/wheelchairRheadless.wrl
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/wrls/kinect.wrl
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/wrls/wheelchairElder.wrl
-- Installing: /usr/local/share/asibot/contexts/ravebot/models/wrls/wheelchairRhead.wrl
-- Installing: /usr/local/share/asibot/contexts/cartesianServer/cartesianServer.ini
[... cut ...]
-- Installing: /usr/local/share/asibot/contexts/cartesianServer/cartesianServerPXA270.ini
-- Installing: /usr/local/lib/libTestRaveBotLib.a
-- Installing: /usr/local/lib/libCartesianClientLib.a
-- Installing: /usr/local/lib/libCartesianServerLib.a
-- Installing: /usr/local/lib/libColorClientLib.a
-- Installing: /usr/local/lib/libRlPlugins.a
-- Installing: /usr/local/lib/libravebot.a
-- Installing: /usr/local/lib/libcartesianbot.a
-- Installing: /usr/local/bin/cartesianServer
-- Removed runtime path from "/usr/local/bin/cartesianServer"

jgvictores, 2016-10-13:
¿Qué distribución estás utilizando (Ubuntu, Mint, versión...)?
¿Algún gestor de ventanas estilo i3?

jgvictores, 2016-11-07:
No se está instalando: `/usr/local/share/yarp/config/path.d/asibot.ini`
Esto normalmente se debe a que existe una variable `ASIBOT_ROOT` o similar.
Please copy+paste results of:
`echo $ASIBOT_DIR`

jgvictores, 2017-11-07:
Closing due to inactivity.

closed 2022-05-25 by PeterBowman