roboticslab-uc3m/asibot-main #9: enable webInterface

DennysMS, 2016-07-14:
No puedo habilitar ni instalar la app de webInterface de ASIBOT.
El texto dice lo siguiente "The module is compiled when ENABLE_webInterface is activated (default: OFF)" pero no lo puedo habilitar. No se si tengo que hacer algo antes.
He seguido el siguiente tutorial "http://robots.uc3m.es/dox-asibot-main/install_on_ubuntu.html" pero ahora quiero utilizar la webInterface pero no puedo.

Gracias de antemano.

DennysMS, 2016-07-14:
También he intentado seguir este tutorial: http://robots.uc3m.es/dox-asibot-main/group__asibot__webInterface.html Pero creo que faltan comandos ya que al introducir el comando "make install_applications" me sale el siguiente mensaje: "make: *** No hay ninguna regla para construir el objetivo «install_applications».  Alto."

jgvictores, 2016-07-14:
Pequeño disclaimer: la repo asibot-main en general la hemos mantenido poco últimamente. Por ejemplo, make install_applications ya no es necesario.
Otra cosa, por favor copia siempre el output del make (desde arriba). Es donde se suelen leer los errores de compilación.
En la carpeta build, haz:
`cmake .. -DENABLE_webInterface=ON`
y continúa con el make, etc.

DennysMS, 2016-07-15:
He intentado lo que me has sugerido, pero al continuar con el make me salen unos errores. 

Al ejecutar "cmake .. -DENABLE_webInterface=ON" me sale lo siguiente, que yo creo que no tiene ningún problema:
dennys@Dennys-pc:~/asibot-main/build$ cmake .. -DENABLE_webInterface=ON
-- Libraries go to /home/dennys/asibot-main/build/lib
-- Executables go to /home/dennys/asibot-main/build/bin
-- Setting up installation of asibot.ini to /usr/local/share/yarp/config/path.d folder.
-- starting plugin library: RlPlugins
--  +++ plugin ravebot, ENABLE_ravebot is set
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   iostreams
--   python
--   thread
--   system
--  +++ plugin cartesianbot, ENABLE_cartesianbot is set
--  +++ plugin kdlbot, SKIP_kdlbot is set
--  +++ plugin skymegabot, SKIP_skymegabot is set
-- ending plugin library: RlPlugins
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dennys/asibot-main/build

Pero al proseguir con el make, salen dos errores:
dennys@Dennys-pc:~/asibot-main/build$ make -j3; sudo make install; cd ../..
[ 41%] [ 41%] [ 41%] Built target TestRaveBotLib
Built target CartesianClientLib
Built target TinyMath
[ 41%] Built target ColorClientLib
[ 41%] Built target TaskLib
[ 41%] Built target CartesianServerLib
[ 41%] [ 41%] Building CXX object src/libraries/RlPlugins/ravebot/CMakeFiles/ravebot.dir/RateThreadImpl.cpp.o
Building CXX object src/libraries/RlPlugins/ravebot/CMakeFiles/ravebot.dir/IDeviceImpl.cpp.o
[ 41%] Built target cartesianbot
[ 41%] Building CXX object src/libraries/RlPlugins/ravebot/CMakeFiles/ravebot.dir/IControlLimitsImpl.cpp.o
/home/dennys/asibot-main/src/libraries/RlPlugins/ravebot/RateThreadImpl.cpp: In member function ‘virtual void RaveBot::run()’:
/home/dennys/asibot-main/src/libraries/RlPlugins/ravebot/RateThreadImpl.cpp:58:34: warning: ‘OpenRAVE::KinBody::Link::GeometryInfo’ is deprecated (declared at /usr/local/include/openrave-0.9/openrave/kinbody.h:239) [-Wdeprecated-declarations]
         std::list<KinBody::Link::GeometryInfo> drawnInfoList;
                                  ^
/home/dennys/asibot-main/src/libraries/RlPlugins/ravebot/RateThreadImpl.cpp:59:37: warning: ‘OpenRAVE::KinBody::Link::GeometryInfo’ is deprecated (declared at /usr/local/include/openrave-0.9/openrave/kinbody.h:239) [-Wdeprecated-declarations]
         KinBody::Link::GeometryInfo drawnInfo;
                                     ^
/home/dennys/asibot-main/src/libraries/RlPlugins/ravebot/RateThreadImpl.cpp:60:42: warning: ‘OpenRAVE::KinBody::Link::GeomSphere’ is deprecated (declared at /usr/local/include/openrave-0.9/openrave/kinbody.h:245) [-Wdeprecated-declarations]
         drawnInfo._type = KinBody::Link::GeomSphere;
                                          ^
In file included from /usr/include/boost/shared_ptr.hpp:17:0,
                 from /usr/local/include/openrave-0.9/openrave/openrave.h:72,
                 from /usr/local/include/openrave-0.9/openrave-core.h:22,
                 from /home/dennys/asibot-main/src/libraries/RlPlugins/ravebot/RaveBot.h:13,
                 from /home/dennys/asibot-main/src/libraries/RlPlugins/ravebot/IDeviceImpl.cpp:3:
/usr/include/boost/smart_ptr/shared_ptr.hpp: In instantiation of ‘boost::shared_ptr<X> boost::dynamic_pointer_cast(const boost::shared_ptr<U>&) [with T = OpenRAVE::SensorBase::CameraGeomData; U = const OpenRAVE::SensorBase::SensorGeometry]’:
/home/dennys/asibot-main/src/libraries/RlPlugins/ravebot/IDeviceImpl.cpp:258:190:   required from here
/usr/include/boost/smart_ptr/shared_ptr.hpp:810:5: error: cannot dynamic_cast ‘0u’ (of type ‘const class OpenRAVE::SensorBase::SensorGeometry*’) to type ‘class OpenRAVE::SensorBase::CameraGeomData*’ (conversion casts away constness)
     (void) dynamic_cast< T* >( static_cast< U* >( 0 ) );
     ^
/usr/include/boost/smart_ptr/shared_ptr.hpp:814:41: error: cannot dynamic_cast ‘(& r)->boost::shared_ptr<T>::get<const OpenRAVE::SensorBase::SensorGeometry>()’ (of type ‘boost::shared_ptr<const OpenRAVE::SensorBase::SensorGeometry>::element_type* {aka const class OpenRAVE::SensorBase::SensorGeometry*}’) to type ‘E* {aka class OpenRAVE::SensorBase::CameraGeomData*}’ (conversion casts away constness)
     E * p = dynamic_cast< E* >( r.get() );
                                         ^
[ 41%] Building CXX object src/libraries/RlPlugins/ravebot/CMakeFiles/ravebot.dir/SharedArea.cpp.o
[ 41%] Building CXX object src/libraries/RlPlugins/ravebot/CMakeFiles/ravebot.dir/WorldRpcResponder.cpp.o
make[2]: *** [src/libraries/RlPlugins/ravebot/CMakeFiles/ravebot.dir/IDeviceImpl.cpp.o] Error 1
make[2]: *** Se espera a que terminen otras tareas....
/home/dennys/asibot-main/src/libraries/RlPlugins/ravebot/WorldRpcResponder.cpp: In member function ‘virtual bool WorldRpcResponder::read(yarp::os::ConnectionReader&)’:
/home/dennys/asibot-main/src/libraries/RlPlugins/ravebot/WorldRpcResponder.cpp:93:42: warning: ‘OpenRAVE::KinBody::Link::GeometryInfo’ is deprecated (declared at /usr/local/include/openrave-0.9/openrave/kinbody.h:239) [-Wdeprecated-declarations]
                 std::list<KinBody::Link::GeometryInfo> scylInfoList;
                                          ^
/home/dennys/asibot-main/src/libraries/RlPlugins/ravebot/WorldRpcResponder.cpp:94:45: warning: ‘OpenRAVE::KinBody::Link::GeometryInfo’ is deprecated (declared at /usr/local/include/openrave-0.9/openrave/kinbody.h:239) [-Wdeprecated-declarations]
                 KinBody::Link::GeometryInfo scylInfo;
[... cut ...]
/usr/include/boost/smart_ptr/shared_ptr.hpp:814:41: error: cannot dynamic_cast ‘(& r)->boost::shared_ptr<T>::get<const OpenRAVE::SensorBase::SensorGeometry>()’ (of type ‘boost::shared_ptr<const OpenRAVE::SensorBase::SensorGeometry>::element_type* {aka const class OpenRAVE::SensorBase::SensorGeometry*}’) to type ‘E* {aka class OpenRAVE::SensorBase::CameraGeomData*}’ (conversion casts away constness)
     E * p = dynamic_cast< E* >( r.get() );
                                         ^
make[2]: *** [src/libraries/RlPlugins/ravebot/CMakeFiles/ravebot.dir/IDeviceImpl.cpp.o] Error 1
make[1]: *** [src/libraries/RlPlugins/ravebot/CMakeFiles/ravebot.dir/all] Error 2
make: *** [all] Error 2

jgvictores, 2016-07-17:
```bash
cd asibot-main
git pull
git checkout develop
git pull
cd build
cmake ..
make
sudo make install
```

DennysMS, 2016-07-20:
Con los comandos que me has dado ya se ha instalado webInterface, pero al ejecutar "bin/webInterface" no se abre el simulador ni me deja acceder a la interfaz web a través del navegador, tal y como sucede en el video "https://www.youtube.com/watch?v=9B0tVJsg-0I&feature=youtu.be"

Lo que me sale al introducir el comando bin/webInterface es:

dennys@Dennys-pc:~/asibot-main/build$ bin/webInterface 
||| clearing context
||| adding context [webInterface/conf]
||| configuring
||| default config file specified as webInterface.ini
||| checking [/home/dennys/asibot-main/build/webInterface.ini] (pwd)
||| checking [/home/dennys/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/dennys/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/ubuntu/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/asibot/robots/default] (robot yarp.d)
||| checking [/usr/local/share/teo/robots/default] (robot yarp.d)
||| checking [/home/dennys/.config/yarp/contexts/webInterface/conf] (context YARP_CONFIG_HOME)
||| checking [/home/dennys/.local/share/yarp/contexts/webInterface/conf] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/webInterface/conf] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/webInterface/conf] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/webInterface/conf] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/contexts/webInterface/conf] (context YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/contexts/webInterface/conf] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/contexts/webInterface/conf] (context YARP_DATA_DIRS)
||| checking [/usr/share/yarp/contexts/webInterface/conf] (context YARP_DATA_DIRS)
||| checking [/usr/local/share/asibot/contexts/webInterface/conf] (context yarp.d)
||| checking [/usr/local/share/teo/contexts/webInterface/conf] (context yarp.d)
||| checking [/home/dennys/.config/yarp/webInterface.ini] (YARP_CONFIG_HOME)
||| checking [/home/dennys/.local/share/yarp/webInterface.ini] (YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/webInterface.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/webInterface.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/webInterface.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/share/ubuntu/yarp/webInterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/share/gnome/yarp/webInterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/webInterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/share/yarp/webInterface.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/share/asibot/webInterface.ini] (yarp.d)
||| checking [/usr/local/share/teo/webInterface.ini] (yarp.d)
||| did not find webInterface.ini
yarp: cannot read from webInterface.ini
Run "webInterface --help" for options.
webInterface checking for yarp network... [ok]
--------------------------------------------------------------
WebInterface using period: 5.000000, resources: robots.uc3m.es/webInterface/html.
WebInterface using webIp: localhost, webPort: 8080.
||| WARNING  Could not create /home/dennys/.local/share/yarp/contexts/webInterface/conf directory
WebInterface using htmlPath: /home/dennys/.local/share/yarp/contexts/webInterface/conf/../html/
||| WARNING  Could not create /home/dennys/.local/share/yarp/contexts/webInterface/conf directory
WebInterface using userPath: /home/dennys/.local/share/yarp/contexts/webInterface/conf/../user/
WebInterface using resourcePath: http://robots.uc3m.es/webInterface/html/
--------------------------------------------------------------
yarp: Port /web active at /localhost:8080
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Server running, visit: http://localhost:8080/index
Request: index (REQUEST_URI index)
filePath: /home/dennys/.local/share/yarp/contexts/webInterface/conf/../html/index.html

jgvictores, 2016-07-21:
En #10 he actualizado webInterface para que encuentre los recursos, pero falta por actualizar el simulador.

jgvictores, 2016-07-21:
```bash
cd asibot-main
git pull  # supongo que sigues en el branch "develop"
cd build
cmake ..
make
sudo make install
```
Lanzando una instancia de `cartesianServer` en otra ventana (es el que invoca al simulador), he podido conectar la interfaz web con el simulador. Prueba y me cuentas.

DennysMS, 2016-07-26:
Lo he probado y si que establece conexión entre la interfaz y el simulador. Pero en cuanto se mueve una articulación desde la interfaz, se empieza a descolocar todo el brazo robótico y no se a que se puede deber eso.
Aparte de esto, el brazo robótico esta muy cerca del mueble de la cocina. Se podría mover todo el brazo robótico a la mitad del entorno, si es así como se podría hacer.

DennysMS, 2016-08-31:
Como al mover el robot desde la interfaz web se desancla o desensambla, he intentado moverlo directamente desde el simulador y desde la terminal.
Al intentar moverlo desde el simulador es como que quiere moverse pero no consigue hacerlo y vuelve a su posición de partida. Mientras que si lo intento mover desde la terminal ejecutando 
`yarp rpc /ravebot/cartesianServer/rpc:i` 
y posteriormente 
`[movl] (.1 .3 .8 90 0)` 
en la terminal donde he lanzado la instancia de `cartesianServer` me sale el siguiente error de ravebot:
```
[RaveBot] fail: RaveBot will not velocityMove as not in velocityMode
[CartesianBot] warning: FAILED ON VELOCITY MOVE!!!
```
a pesar de que en la terminal donde ejecuto el movimiento me sale lo siguiente:
```
[movl] (.1 .3 .8 90 0) 
Response: [ok]
```

jgvictores, 2016-09-01:
Procedo a cerrar esta issue, puesto que webInterface está actualizado y se conecta.

Por favor abre otras issues separados por temática, como podrían ser issues diferentes para estabilidad de motor físico vs problemas en movimiento cartesiano movl.

closed 2016-09-01 by jgvictores