roboticslab-uc3m/asibot-main #11: Estabilidad de motor fisico en simulación

DennysMS, 2016-09-01:
Al intentar mover el robot desde la interfaz web, este se desancla o desensambla y las articulaciones del robot empiezan a moverse libremente por el entorno. 
He intentado moverlo directamente desde el simulador pero es como que quiere moverse pero no consigue hacerlo y vuelve a su posición de partida, es decir, el motor no se estabiliza cuando se intenta un movimiento ya sea desde la interfaz o desde el propio simulador.

jgvictores, 2016-09-08:
Los motores físicos de simulación tienden a ser muy inestables. Recomiendo la desactivación del motor físico. De hecho, en https://github.com/roboticslab-uc3m/asibot-main/blob/develop/src/libraries/RlPlugins/ravebot/RaveBot.h se ve `DEFAULT_PHYSICS "none"`. Si no lo estás activando por línea de comandos, revisa que no sea el `xml` del entorno el que lo está activando.

DennysMS, 2016-09-08:
Como bien decías, era el xml del entorno que activaba los motores físicos. 
Simplemente he comentado la parte de `physicsengine` y en simulación ya funciona correctamente.
Gracias.

jgvictores, 2016-09-10:
Sí, la idea es esa: cambiar parámetros a través de línea de comandos o en ficheros de configuración (`.ini`, `.xml`,...) no está mal porque precisamente están para eso y no hay que recompilar.
Cierro esta issue, pero de momento dejo abierta la #12 porque en esa la única solución propuesta afecta al código fuente y hay que recompilar.

closed 2016-09-10 by jgvictores