roboticslab-uc3m/asibot-main #12: Problemas con movimiento cartesiano

DennysMS, 2016-09-01:
Al intentar mover el robot desde la terminal (2) ejecutando los siguientes comandos:
`yarp rpc /ravebot/cartesianServer/rpc:i`
`[movl] (.1 .3 .8 90 0)`

en la terminal (1) donde he lanzado la instancia de `cartesianServer` me sale el siguiente error de ravebot:

```
[RaveBot] fail: RaveBot will not velocityMove as not in velocityMode
[CartesianBot] warning: FAILED ON VELOCITY MOVE!!!
```

a pesar de que en la terminal (2) donde se ejecuta el movimiento sale como que todo esta correcto

```
[movl] (.1 .3 .8 90 0) 
Response: [ok]
```

DennysMS, 2016-09-01:
Problema solucionado. Había que cambiar el control de posición a velocidad `DEFAULT_MODE_POS_VEL = 1`
Pero al moverse el robot se desacoplan las articulaciones, es decir, se tiene el mismo problema que en #11

jgvictores, 2016-09-08:
En realidad no habría que tocar los valores por defecto. 'movl' utiliza el modo velocidad, y es posible que no se esté cambiando al modo de velocidad debido a la utilización de una función obsoleta.

PeterBowman, 2016-10-29:
El problema original debería haber quedado resuelto con 067d271, llevo un tiempo realizando simulaciones con alternancia de modos y no he vuelto a observar ese error. ¿Podrías confirmarlo, @DennysMS, con la versión más reciente del repositorio en develop? Si aún tienes aquella modificación en el código, [esto](http://stackoverflow.com/a/7147320) debería deshacerla fácilmente.

Esta primera aproximación consiguió que ControlBoardWrapper reconociera la función setVelocityMode, ahora miembro de IControlMode con una signatura diferente. En 5517694, las llamadas a funciones obsoletas fueron reemplazadas para evitar problemas similares en el futuro, y en la rama control-mode estoy finalizando ahora la transición a IControlMode2.

closed 2016-11-07 by PeterBowman