roboticslab-uc3m/asibot-main #13: Creación de programa en web nterface

DennysMS, 2016-09-09:
Tengo problemas al intentar hacer un programa en la interfaz web. Guarda los puntos en la pestaña _joint_ pero al ir a la pestaña _program_ dice que no se ha guardado ningún punto, a pesar de que he creado un programa tal y como sale en el tutorial https://www.youtube.com/watch?v=9B0tVJsg-0I

jgvictores, 2016-09-10:
Por un lado hay pestañas de movimiento, pero el punto se captura con el botón de _Capture Point_ que está parte superior derecha en la pestaña _program_, a modo 'estoy conforme con esta posición que estoy viendo'.

DennysMS, 2016-09-11:
Es verdad que en la pestaña de _program_ hay una parte donde pone _captured points_ pero no hay ningún botón, por eso yo guardaba los puntos desde la pestaña _joint_ pensando que se guardaría desde allí también pero ya veo que no es así. 
Entonces esa es mi duda, no hay botón para capturar puntos desde la pestaña _program_, por tanto no se puede hacer un programa.

jgvictores, 2016-09-11:
Si te fijas en [2:51 del vídeo](https://youtu.be/9B0tVJsg-0I?t=2m51s), hace click en el botón de _Capture point_ (no el texto, sino la imagen del punto rojo estilo 'grabar' justo debajo). De ahí además se le pide confirmación en una pequeña ventana emergente.

 1. ¿Tienes el botón rojo en pantalla? (lo he señalado con una flecha rosa sobre un pantallazo)
 2. ¿Te sale la ventana emergente? (sólo se me ocurre que tengas bloqueado cualquier Alerta de javascript en el navegador; en todo caso, suele saltar un aviso de intento de ventana emergente)

[image]

DennysMS, 2016-09-12:
El problema es que no me sale el botón rojo en pantalla e incluso he visto el código de _program.html_ y en el apartado de _POINTS_ está vacío, aunque he de reconocer que no tengo mucha idea de html ni javascript.
Adjunto un pantallazo de lo que me sale en la pestaña _program_

[image]

jgvictores, 2016-09-12:
Tienes razón, el vídeo está desactualizado. En versiones más recientes de `webInterface` los puntos se guardan desde las pestañas de movimiento. Con c6ca734bad1e195d9e2d994d8cfed58688f49402 he arreglado lo de que aparezcan los puntos en la pantalla de programa (puedes hacer el clásico `git pull`, recompilar y volver a instalar), pero todavía hay problemas de creación de programas debido a rutas y a permisos. Iré arreglando estos problemas y te mantendré al tanto.

DennysMS, 2016-09-12:
Lo acabo de probar y si, es verdad que ya se guardan los puntos y también me deja crear un programa con dichos puntos. Pero ahora la cuestión es ejecutar ese programa, que supongo que se hace desde la pestaña _asiggner_. 
Una vez creado y guardado el programa, me dirijo a la pestaña _assigner_, elijo el programa creado y le doy al botón _Assign to Task_, pero no sucede nada, ni sale una ventana emergente.
De todas formas estaré esperando actualizaciones, gracias.

PeterBowman, 2016-10-04:
@DennysMS _Assign to task_ requiere haber especificado un programa, un comando de voz y un icono, pero únicamente muestra una alerta si olvidas seleccionar alguno de los iconos. El comando de voz es lo que más guerra da porque no funciona la escucha de eventos en JavaScript para el cambio de contenido en el &lt;input&gt; de la pestaña _Speech_. Una alternativa es crear manualmente el fichero _words.ini_ en tu _userPath_ (en mi caso _~/.local/share/yarp/contexts/webInterface/_) y añadir allí tus comandos, uno por línea.
@jgvictores en PeterBowman/asibot-main@e2313c7 he corregido el _handler_ para que guarde correctamente los comandos de voz y he añadido también un par de ventanas emergentes para que el usuario no se pierda si olvida introducir algún parámetro.

jgvictores, 2016-10-04:
@PeterBowman Creo que aún hay rutas rotas (esto se debe a cambios en todo lo de _context_ de YARP), pero buenas mejoras! Un PR sería muy bienvenido. :smile:

PeterBowman, 2016-10-04:
@jgvictores el único problema con las rutas que había notado es que no me encontraba los scripts de /share/../user, en particular _template.py_, que carga un trozo de código necesario para ejecutar las tareas guardadas (pestaña _Launcher_). Esto debería estar ya resuelto con PeterBowman/asibot-main@7a58b96, aunque sería preferible buscar otra solución. Por lo que veo, WebResponder espera que tanto _template.py_ como _AsibotPy.py_  estén ambos en _userPath_, luego puede que lo óptimo sea copiarlos durante la instalación. La solución actual requiere que _AsibotPy.py_ esté incluido en la variable _$PYTHONPATH_.

jgvictores, 2016-10-06:
En cuanto a `AsibotPy.py`, etc... lo ideal sería crear un `setup.py`. CMake sería el encargado de invocar `python setup.py install`.

PeterBowman, 2016-11-14:
> En cuanto a AsibotPy.py, etc... lo ideal sería crear un setup.py. CMake sería el encargado de invocar python setup.py install.

@jgvictores esto ya estaría listo en la rama `setup-py`, CMake creará el módulo `AsibotPy` (llamando a `AsibotPy.py`) durante la instalación. ¿Hay algo más que quede por hacer en esta issue? ¿Qué te parecería incluir `example/python/CartesianClient.py` en ese mismo módulo para evitar la necesidad de configurar `ASIBOT_ROOT`? Las únicas referencias restantes a esta variable (según tengo entendido, estaba siendo eliminada progresivamente) serían las de la documentación y una más en [example/exportable/ExternalModules.cmake.template](https://github.com/roboticslab-uc3m/asibot-main/blob/705e903/example/exportable/ExternalModules.cmake.template), que no parece que se esté usando. Otra opción es olvidarnos por completo de [`AsibotPy.py`](https://github.com/roboticslab-uc3m/asibot-main/blob/705e903/share/webInterface/user/AsibotPy.py), instalar un módulo para `CartesianClient` e importarlo directamente en [`template.py`](https://github.com/roboticslab-uc3m/asibot-main/blob/705e903/share/webInterface/user/template.py) junto con YARP y la comprobación de conectividad.

jgvictores, 2016-11-24:
>  ¿Qué te parecería incluir example/python/CartesianClient.py en ese mismo módulo para evitar la necesidad de configurar `ASIBOT_ROOT`?

Me parece bien. De hecho:
1. `example/exportable` es una cosa obsoleta desde que existen las shared libraries de YARP.
2. En realidad CartesianClient.py (además de CartesianClient.java) deberían desaparecer a corto plazo, siendo reemplazados por un enfoque más parecido al de ColorClient (nota: todos estos también desaparecerán... :skull_and_crossbones: :muscle: :smile: :muscle: ).

En conclusión: refractorizaremos mucho, pero de momento mover el código de `example/python/CartesianClient.py` a `AsibotPy.py` es una solución muy elegante. Por favor, ¡adelante!

PR #15 (2016-10-06): Some work on the 'Assigner' tab in webInterface
Resolves #13.
PR #25 (2016-11-25): Setup AsibotPy module at installation
Should close #13.
closed 2016-11-25 by jgvictores