roboticslab-uc3m/asibot-main #16: Problemas para ejecutar la tarea en webInterface y el movimiento cartesiano.

serna92, 2016-10-10:
Buenas tardes,

Tengo problemas al ejecutar una tarea en webInterface desde la pestaña launcher.
Cuando ejecuto la tarea me dice que el módulo AsibotPy no existe para ser importado. Agregué el archivo en el userpath (/home/miguel/.local/share/yarp/contexts/webInterface) y pasaba a decirme que no existía el módulo Yarp el cual no llegue a encontrar.

Creo que tras la instalación del webInterface, no se me están creando archivos o, como el userpath, parece que toma una ruta incorrecta donde crearlos.

Adjunto una foto tanto del userpath que se creó y de la carpeta que se instaló en "/usr/local/share/asibot/contexts/webInterface".

También, aun existe el problema en el cartesianServer de que no se alterna entre movimiento cartesiano y movimiento articular cuando cartesianServer esta funcionando, o se lanza cartesianServer inicialmente con un tipo de movimiento o se lanza con el otro.

Gracias por adelantado y disculpa las molestias.

[image]
[image]
[image]

PeterBowman, 2016-10-10:
La detección de AsibotPy.py es una tarea pendiente de #13. Si no quieres copiar ese archivo a tu carpeta local (/home/miguel/.local/...), puedes añadir la ruta correspondiente a la variable de entorno $PYTHONPATH (asumiendo que ya has instalado Asibot en /usr como indican las capturas de pantalla):

`$ echo 'export PYTHONPATH=$PYTHONPATH:/usr/local/share/asibot/contexts/webInterface/user' >> ~/.bashrc`

El error que comentas podría indicar que no has compilado los bindings de YARP para Python. Si es el caso, sigue las instrucciones de http://www.yarp.it/yarp_swig.html#yarp_swig_python. Asegúrate de compilarlos para Python 2.x, ajustando las variables avanzadas PYTHON_ de CMake según corresponda.

El problema de la falta de alternancia entre modos cartesiano y articular de cartesianServer corresponde a #12. Creo que ya he conseguido identificar el problema; trataré de arreglarlo y subir un parche, o dejar un comentario en esa issue con lo que he averiguado hasta ahora.

serna92, 2016-10-11:
Buenas, volví a ejecutar el programa tras añadir la variable de entorno e instalar los bindings y me apareció un nuevo error.

Request: execute (program fill_glass) (REQUEST_URI "execute?program=fill_glass")
execute fill_glass program.
Traceback (most recent call last):
  File "/home/miguel/.local/share/yarp/contexts/webInterface/fill_glass.py", line 1, in <module>
    from AsibotPy import *
  File "/usr/local/share/asibot/contexts/webInterface/user/AsibotPy.py", line 9, in <module>
    asibot_root = os.environ['ASIBOT_ROOT']
  File "/usr/lib/python2.7/UserDict.py", line 23, in __getitem__
    raise KeyError(key)
KeyError: 'ASIBOT_ROOT'
The value returned was: 256.

¿A qué se debe?

jgvictores, 2016-10-11:
Antiguamente se utilizaba la variable de entorno ([leer acerca de viariables de entorno en general](http://asrob.uc3m.es/index.php/Variables_de_entorno)) ASIBOT_ROOT. Algo así como:
`export ASIBOT_ROOT=$HOME/asibot-main`
antes de lanzar el programa, o:
`$ echo 'export ASIBOT_ROOT=$HOME/asibot-main' >> ~/.bashrc`
como solución más permanente debería ayudar hasta que arregle el código. Por favor prueba y nos comentas.

serna92, 2016-10-11:
Utilicé la solución de usar el comando "echo 'export ASIBOT_ROOT=$HOME/asibot-main' >> ~/.bashrc" y ya se ejecuta la simulación del movimiento del robot.

Por cierto, ¿Cómo se simula el agarre de un objeto en el escenario?.

PeterBowman, 2016-10-13:
@serna92 he probado con los comandos RPC de [este tutorial](http://robots.uc3m.es/dox-asibot-main/group__testRaveBot.html#testRaveBot_interfacing). El siguiente código Python, pegado en la lista de instrucciones de un programa en webInterface (siempre después de la instrucción `from AsibotPy import *`), genera un cubo en el simulador (equivale a `[on terminal 4] world mk sbox .1 .1 .1 1 2 1.5` de dicho tutorial):

```python
rpc = yarp.RpcClient()

rpc.open('/command/ravebot/world')

yarp.Network.connect('/command/ravebot/world', '/ravebot/world')

cmd = yarp.Bottle()

cmd.addString('world')
cmd.addString('mk')
cmd.addString('sbox')
cmd.addDouble(0.1)
cmd.addDouble(0.1)
cmd.addDouble(0.1)
cmd.addDouble(1)
cmd.addDouble(2)
cmd.addDouble(1.5)

res = yarp.Bottle()

rpc.write(cmd, res)
```

jgvictores, 2016-10-13:
Habiendo hecho eso, esa misma caja se podría agarrar con:
```python
cmd2 = yarp.Bottle()

cmd2.addString('world')
cmd2.addString('grab')
cmd2.addString('sbox')
cmd2.addInt(1)
cmd2.addInt(1)

rpc.write(cmd2, res)
```
y soltar con:
```python
cmd3 = yarp.Bottle()

cmd3.addString('world')
cmd3.addString('grab')
cmd3.addString('sbox')
cmd3.addInt(1)
cmd3.addInt(0)

rpc.write(cmd3, res)
```

PeterBowman, 2016-11-28:
Abordaré el tema de las llamadas a `ASIBOT_ROOT` en #26. ¿Hay algo más que quede por hacer aquí?

jgvictores, 2016-11-28:
Yo esto lo cerraría como duplicado de #12.

closed 2022-05-25 by PeterBowman