roboticslab-uc3m/asibot-main #19: Problemas para compilar asibot-main.

serna92, 2016-10-21:
Buenos días,

Al ejecutar el comando make aparece el siguiente error:

[ 25%] Building CXX object src/libraries/RlPlugins/ravebot/CMakeFiles/ravebot.dir/__/__/__/__/generated_code/RlPlugins_add_ravebot.cpp.o
In file included from /usr/local/include/yarp/os/SharedLibraryFactory.h:16:0,
                 from /usr/local/include/yarp/os/SharedLibraryClassFactory.h:13,
                 from /usr/local/include/yarp/os/SharedLibraryClass.h:13,
                 from /home/miguel/asibot-main/build/generated_code/RlPlugins_add_ravebot.cpp:10:
/home/miguel/asibot-main/build/generated_code/RlPlugins_add_ravebot.cpp: In function ‘void* ravebot_create()’:
/home/miguel/asibot-main/build/generated_code/RlPlugins_add_ravebot.cpp:35:38: error: cannot allocate an object of abstract type ‘RaveBot’
 YARP_DEFINE_SHARED_SUBCLASS(ravebot, RaveBot, yarp::dev::DeviceDriver)
                                      ^
In file included from /home/miguel/asibot-main/build/generated_code/RlPlugins_add_ravebot.cpp:12:0:
/home/miguel/asibot-main/src/libraries/RlPlugins/ravebot/RaveBot.h:73:7: note:   because the following virtual functions are pure within ‘RaveBot’:
 class RaveBot : public DeviceDriver, public RateThread, public IPositionControl, public IVelocityControl, public IEncodersTimed, public IControlLimits, public IControlMode {
       ^
In file included from /usr/local/include/yarp/dev/ControlBoardInterfaces.h:26:0,
                 from /home/miguel/asibot-main/src/libraries/RlPlugins/ravebot/RaveBot.h:8,
                 from /home/miguel/asibot-main/build/generated_code/RlPlugins_add_ravebot.cpp:12:
/usr/local/include/yarp/dev/IPositionControl.h:190:18: note: 	virtual bool yarp::dev::IPositionControl::setPositionMode()
     virtual bool setPositionMode()=0;
                  ^
In file included from /usr/local/include/yarp/dev/ControlBoardInterfaces.h:24:0,
                 from /home/miguel/asibot-main/src/libraries/RlPlugins/ravebot/RaveBot.h:8,
                 from /home/miguel/asibot-main/build/generated_code/RlPlugins_add_ravebot.cpp:12:
/usr/local/include/yarp/dev/IVelocityControl.h:135:18: note: 	virtual bool yarp::dev::IVelocityControl::setVelocityMode()
     virtual bool setVelocityMode()=0;
                  ^
In file included from /home/miguel/asibot-main/build/generated_code/RlPlugins_add_ravebot.cpp:11:0:
/usr/local/include/yarp/dev/Drivers.h: In instantiation of ‘yarp::dev::DeviceDriver* yarp::dev::DriverCreatorOf<T>::create() [with T = RaveBot]’:
/home/miguel/asibot-main/build/generated_code/RlPlugins_add_ravebot.cpp:35:1:   required from here
/usr/local/include/yarp/dev/Drivers.h:135:20: error: cannot allocate an object of abstract type ‘RaveBot’
         return new T;
                    ^
In file included from /home/miguel/asibot-main/build/generated_code/RlPlugins_add_ravebot.cpp:12:0:
/home/miguel/asibot-main/src/libraries/RlPlugins/ravebot/RaveBot.h:73:7: note:   since type ‘RaveBot’ has pure virtual functions
 class RaveBot : public DeviceDriver, public RateThread, public IPositionControl, public IVelocityControl, public IEncodersTimed, public IControlLimits, public IControlMode {
       ^
make[2]: *** [src/libraries/RlPlugins/ravebot/CMakeFiles/ravebot.dir/__/__/__/__/generated_code/RlPlugins_add_ravebot.cpp.o] Error 1
make[1]: *** [src/libraries/RlPlugins/ravebot/CMakeFiles/ravebot.dir/all] Error 2
make: *** [all] Error 2

¿Qué debería hacer para solucionarlo?

jgvictores, 2016-10-21:
No compila porque hay funciones de YARP que no están implementadas. Estaban implementadas, pero se borraron en #17 al tratarse de funcionalidades obsoletas. Actualiza YARP:
```bash
cd tu_ruta_al_codigo_fuente_de_yarp
git pull
cd build
cmake ..
make -j3
```
Y prueba de nuevo.

serna92, 2016-10-21:
Ok, ya funcionó.

PeterBowman, 2016-10-21:
En 067d271 (#17) se implementó IControlMode en RaveBot, necesario para funcionar en YARP 2.3.64+, pero el parche en sí requiere una versión posterior (2.3.65+).

@jgvictores: he creado #20 en caso de que fuera preferible ofrecer soporte para versiones más antiguas. De lo contrario, habría que documentar este requisito en alguna parte. ¿Qué te parece?

jgvictores, 2016-10-21:
Back-compatibility forever mientras sea factible :smile:
Ahora aceptaré #20.

jgvictores, 2016-10-21:
@PeterBowman Trabajar con forks está muy bien, pero si lo ves claro, no dudes en trabajar sobre una branch del repo o incluso un hot-fix a develop. BTW, los PR a develop, por favor. La idea es merge a master cuando webInterface están estables. :grinning:

PeterBowman, 2016-10-21:
Muchas gracias, @jgvictores. Sí, al principio dudé y estuve subiendo todo a master, pero desde los últimos commits estoy dejando esa rama a un lado (algunos cambios los subo directamente a develop, otros mediante branches por separado y posterior merge). He usado un PR esta vez porque me parecía una mejor opción para recoger feedback, revisar el código y subir correcciones si hicieran falta. ¿El objetivo es implantar un workflow como en los repos de Teo o algo similar? (PS perdón por el off-topic)

jgvictores, 2016-10-21:
Sí, eso es, seguir el GitFlow como en su día empezamos en los repositorios de teo (véase https://github.com/roboticslab-uc3m/teo-main/issues/16).
Cierro issue!

closed 2016-10-21 by jgvictores