roboticslab-uc3m/asibot-main #21: Clean asibot-main

jgvictores, 2016-10-21:
There are a number of modules that should go elsewhere or become depreciated in favor of advances. This is not mandatory, but we have been working a lot elsewhere!

- [ ] cartesianServer -> https://github.com/roboticslab-uc3m/teo-main/tree/develop/libraries/TeoYarp/BasicCartesianControl which also has a nice server wrapper + client pair in the same tree.
- [ ] colorDepth -> https://github.com/roboticslab-uc3m/teo-head/tree/develop/programs/colorRegionDetection
- [ ] robotDraw -> should be moved to https://github.com/roboticslab-uc3m/xgnitive

...there are some more, I will be updating this issue, we can comment here too!

jgvictores, 2016-11-11:
Updated info on testRaveBot: work at https://github.com/roboticslab-uc3m/openrave-yarp-plugins

PeterBowman, 2016-11-25:
More to come: https://github.com/roboticslab-uc3m/asibot-main/issues/13#issuecomment-262739920.

jgvictores, 2017-04-22:
Moved `pointAtObjectServer` at 0995af2158727327bb9b0753a825ef70a9cd0032, WIP on the current `export-pointAtObjectServer` branch at https://github.com/roboticslab-uc3m/vision.

jgvictores, 2017-05-02:
Edit: Best to move `robotDraw` to [openrave-yarp-plugins](https://github.com/roboticslab-uc3m/openrave-yarp-plugins) rather than `xgnitive`.

jgvictores, 2017-05-03:
Developing replacement for `premultH` on `transCoordsUsingJoints` branch of https://github.com/roboticslab-uc3m/kinematics-dynamics

jgvictores, 2017-05-06:
In fact, `premultH` could be moved to https://github.com/roboticslab-uc3m/kinematics-dynamics as something like `transCoodsUsingH`.

PeterBowman, 2017-08-16:
I've adapted the *wiimoteServer* application to work as a YARP plugin, done at roboticslab-uc3m/yarp-devices@b3d27fc. This is currently implemented as a `yarp::dev::IAnalogSensor` for simplicity/brevity, but should get its own interface and network wrapper sooner or later, with full control of all keys, sensors, rumble device and LEDs. Apart from that, I gave up on CWiiD and switched to the XWiimote library: [Install XWiimote](https://github.com/roboticslab-uc3m/installation-guides/blob/32f402df065c9363dd420fa6cb73efdee59f3ab1/install-xwiimote.md).

PeterBowman, 2018-01-06:
> In fact, premultH could be moved to https://github.com/roboticslab-uc3m/kinematics-dynamics as something like transCoodsUsingH.

@jgvictores `transCoordsUsingJoints` was already written and merged in kinematics-dynamics at the time you posted this, did you mean that it should be renamed? I'm not sure what's left to do here, perhaps we can finally check that box.

jgvictores, 2018-01-06:
The intention is to move and rename. `transCoordsH` and renamed `transCoordsUsingJoints` provide different functionalities.

PeterBowman, 2018-01-06:
Oh, OK. To summarize:

* `transCoordsH`: formerly `premultH`, pre-multiplies a runtime-constant HT matrix (given via YARP configure option). Not done yet.
* `transCoordsUsingJoints`: same as above, but works with online HT (retrieves current joint angles from the robot control board and performs FK). Already done.

Please correct me if I'm wrong.

jgvictores, 2018-01-06:
That's the objective, right!

jgvictores, 2018-01-24:
visionSegmentor:
- Export: 2bba02f558809214f48b3333ed55844aa94e2d3e
- Import: https://github.com/roboticslab-uc3m/vision/commit/ec92007a8174f68eef4afd13d6d5cf78ea220346 and https://github.com/roboticslab-uc3m/vision/commit/2747559c170daa9ea0c9fa6f98f33f8fa19ed781

Will be renamed to colorRegionDetection2D, see https://github.com/roboticslab-uc3m/vision/issues/75

PeterBowman, 2018-01-26:
premultH (re)moved at roboticslab-uc3m/kinematics-dynamics#137.

jgvictores, 2018-02-03:
robotDraw moved to https://github.com/roboticslab-uc3m/openrave-yarp-plugins/tree/d7c946bfa51b3bf012b5eab9962d427af3e58d8b/example/cpp/robotDraw at bf2b2cd55ccf5060f0acfa2fd42bbded415af47a

jgvictores, 2018-02-03:
kinectPxToReal 1bc0569a5725ece3cc65e1540c7b3294817af2ff -> https://github.com/roboticslab-uc3m/vision/commit/c1529c4a1d282aff15fe395a6dd78524db67b985

jgvictores, 2018-02-05:
Marked asibotTasks and taskServer, will be tracked on #40

jgvictores, 2018-02-05:
mv matlab/visualServo out of repo done at 81f9a4075509629a639e4800b80e153fc1cf1be1.

Closing issue!

closed 2018-02-05 by jgvictores

jgvictores, 2018-02-05:
> mv matlab/visualServo out of repo done at 81f9a40.

Created https://github.com/roboticslab-uc3m/matlab-visual-servo for this legacy code, archived.