roboticslab-uc3m/asibot-main #24: Use std::string instead of yarp::os::ConstString

PeterBowman, 2016-11-25:
As stated in its documentation, it's:

> A string with almost the same api as std::string. It adds a hash function for compatibility with certain versions of the ACE library.

In fact, `yarp::os::ConstString` seems to be a mere wrapper around `std::string`. TEO code happens to use this class scarcely, putting much more stress on the latter.

PeterBowman, 2016-11-25:
Putting this on hold in prevision of potentially overlapping work at #21.

jgvictores, 2016-11-25:
I've updated this preference and others on http://robots.uc3m.es but started https://github.com/roboticslab-uc3m/QA/issues/4 to enhance documenting best or recommended practices.

PeterBowman, 2017-05-01:
I'm going to close this as invalid. Several coding style recommendations were written down at the new [best-practices](https://github.com/roboticslab-uc3m/best-practices) repo and I think this one doesn't deserve a separate issue anymore, given most C++ code will get eventually removed or properly modernized anyway as a result of the ongoing reorganization process.

closed 2022-05-25 by PeterBowman