roboticslab-uc3m/asibot-main #28: Clean asibot-main libraries

jgvictores, 2017-02-10:
There are a number of libraries that should go elsewhere or become deprecated in favor of advances. This is not mandatory, but we have been working a lot elsewhere
- [ ] CartesianClientLib
- [ ] CartesianServerLib
- [ ] ColorClientLib
- [ ] RavePlugins
- RlPlugins
  - [ ] cartesianbot
  - [ ] kdlbot
  - [ ] ravebot
  - [ ] skymegabot
- [ ] TaskLib
- [ ] TestRaveBotLib
- [ ] TinyMath
- [ ] TrajGen

PeterBowman, 2017-05-01:
Note: TinyMath is still called by TaskLib.

jgvictores, 2017-10-31:
`ravebot` was never actually migrated, but I'd say that with https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/38, it is definitely superseeded by YarpOpenraveControlboard.

jgvictores, 2017-10-31:
Well, maybe not completely superseeded until https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/29 and https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/39 are done.

jgvictores, 2017-11-29:
https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/29 is now closed, `YarpOpenraveControlboard` is an efficient re-implementation of `ravebot`.

As for https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/39, it just provides a couple of checks and parameters, will be soon closed.

jgvictores, 2018-02-05:
Marking `TaskLib` too (which was the only lib missing), of which progress will be tracked at #40.

Closing issue!

closed 2018-02-05 by jgvictores