roboticslab-uc3m/asibot-main #31: Delete backup data and wipe heavy files from commit history

jgvictores, 2017-04-22:
We should be removing old big files (`*.iv`, `*.wrl`) from this repository.

Issue discovered while looking at https://github.com/roboticslab-uc3m/asibot-openrave-models/issues/1 on heavy asibot models.

jgvictores, 2017-04-22:
@PeterBowman Safe to do this some time tomorrow?

PeterBowman, 2017-04-22:
@jgvictores sure!

jgvictores, 2017-04-24:
Again using https://rtyley.github.io/bfg-repo-cleaner/

```bash
git clone --mirror https://github.com/roboticslab-uc3m/asibot-main
java -jar ../../src/bfg-1.12.15.jar --delete-files '{*.wrl,*.iv}' asibot-main.git
```

Filename           |       Git id       
--------------------|----------------------------------------------
cocina.iv               | e69f173a (1.2 MB)                       
docking.iv              | 8f086087 (1.2 MB)                      
dstation.iv             | 00d5f345 (719.0 KB)                     
elmacho.iv              | 3885fc2b (702.7 KB)                     
ensam01sn.iv            | c53177a7 (3.5 MB)                       
ensam02sn.iv            | f2027f28 (3.7 MB)                       
ensam03sn.iv            | ca93b72e (5.9 MB)                      
ensam04sn.iv            | 4cce3ac4 (5.8 MB)                       
kinect.wrl              | 0038ee4c (271.5 KB), 12631b95 (126.2 KB)
lata.iv                 | ed6b4c3f (12.4 MB)                      
machosn.iv              | b02a23c3 (1.1 MB)                       
silla.iv                | 7e25bc85 (6.1 MB)                       
wheelchair.iv           | 7e25bc85 (6.1 MB)                       
wheelchairElder.wrl     | 504cd794 (11.9 MB)                     
wheelchairRhead.wrl     | e661a183 (157.8 KB)           
wheelchairRheadless.wrl | 3ecd182f (30.0 MB)  

```bash
cd asibot-main.git
git reflog expire --expire=now --all && git gc --prune=now --aggressive
git push
```

Mmm I think this was already done at some time. Was `13.1 M`, remains at `13.1 M`. :sweat_smile:

Would close this issue as invalid, but `push` were made, so must recommend fresh clone of repo to avoid conflicts.

jgvictores, 2017-04-24:
@roboticslab-uc3m/asibot Please delete your old copy of [asibot-main](https://github.com/roboticslab-uc3m/asibot-main) and fresh clone!!!

closed 2017-04-24 by jgvictores

PeterBowman, 2017-04-24:
:+1: 

I think it actually did something as the `share/ravebot/models/ivs/` directory is no longer there at e4546c0.