roboticslab-uc3m/asibot-main #33: Adding a Kinbody after an environment has been loaded by cartesianserver or testravebot.

serna92, 2017-05-19:
Hello,

I need to add an object depending on the task I want to simulate with the robot asibot.

As you can see the environment and the objects that I use are in different files:
- [Environment](https://github.com/serna92/AsibotSimulation/blob/replaceCartesianServer/entornoAsibot/asibot_kitchen.env.xml)
- [RedCan](https://github.com/serna92/AsibotSimulation/blob/replaceCartesianServer/entornoAsibot/redCan.kinbody.xml)
- [Glass](https://github.com/serna92/AsibotSimulation/blob/replaceCartesianServer/entornoAsibot/glass.kinbody.xml)
- [Bottle](https://github.com/serna92/AsibotSimulation/blob/replaceCartesianServer/entornoAsibot/bottle.kinbody.xml)
- [Dish](https://github.com/serna92/AsibotSimulation/blob/replaceCartesianServer/entornoAsibot/dish.kinbody.xml)

Another way is to have 3 environment (1 per implemented task) and launch cartesianserver / testravebot with this environment as argument. How can I launch them inside a python file instead of command line? 

Thanks and sorry by the inconveniences.

jgvictores, 2017-05-19:
"Adding a Kinbody after an environment has been loaded by cartesianserver or testravebot" would be implemented in a `WorldRpcResponder` plugin that should be ported/created as part of https://github.com/roboticslab-uc3m/openrave-yarp-plugins/issues/27

jgvictores, 2017-05-19:
PS: You could actually implement that as part of the current [WorldRpcResponder.cpp](https://github.com/roboticslab-uc3m/asibot-main/blob/ec0b387fe3ea7eed6ccd0087f296dd0172193730/src/libraries/RlPlugins/ravebot/WorldRpcResponder.cpp) as part of the `mk` command.

serna92, 2017-05-19:
I added this in `mk command` on `WorldRpcResponder`:

```
            } else if (in.get(2).asString() == "obj") {
                { // lock the environment!           
                OpenRAVE::EnvironmentMutex::scoped_lock lock(pEnv->GetMutex());
                KinBodyPtr objKinBodyPtr = RaveCreateKinBody(pEnv,"");
                pEnv->ReadKinBodyXMLFile(objKinBodyPtr, in.get(3).asString().c_str());
                pEnv->Add(objKinBodyPtr,true);
                }  // the environment is not locked anymore
                out.addVocab(VOCAB_OK);
```

and after reinstall asibot-main again and execute this in my python script it worked as expected:

```
cmd3 = yarp.Bottle()

   cmd3.addString('world')
   cmd3.addString('mk')
   cmd3.addString('obj')
   cmd3.addString('AsibotSimulation/entornoAsibot/redCan.kinbody.xml')

   rpc.write(cmd3, res)
```

jgvictores, 2017-05-20:
> Should it be commited?

Yes, that looks good (you should always commit, see https://github.com/roboticslab-uc3m/best-practices!).

Please commit the lines of each file on branches (or branch of fork if you do not have permissions) to later create Pull Request(s) (PR) or merge directly if functional across PCs.

PeterBowman, 2017-07-10:
Looks like #34 solved this, feel free to re-open if needed.

closed 2017-07-10 by PeterBowman