roboticslab-uc3m/asibot-main #35: Alternative to simCart.wait() for `pos.positionMove`

serna92, 2017-05-23:
Hi,

When I use `pos.positionMove`, [my pogram](https://github.com/serna92/AsibotSimulation/blob/replaceCartesianServer/programas/pick_can.py#L73) continue running to the next line which is a `simCart.movl()`. Then both robot movements are executed at the same time. Im using `yarp.Time.delay(20)` but is too variable since some movements last 18 seconds but others can last 25 seconds. I would like to make it wait till the target position was reached.

There is a code similar to `simCart.wait()` but for `pos.positionMove`?

PeterBowman, 2017-05-23:
`simCart.wait()` awaits a reply from the RPC server at [AsibotPy.py#L92](https://github.com/roboticslab-uc3m/asibot-main/blob/f9fb22d/share/webInterface/user/AsibotPy.py#L92), thus blocking your thread of execution until a response is available. To makes this work with the controlboard interfaces (e.g. `IPositionControl`), you'd have to implement the wait loop in your own code; see how this is implemented in [xRpcResponder.cpp](https://github.com/roboticslab-uc3m/asibot-main/blob/f9fb22d/src/libraries/CartesianServerLib/xRpcResponder.cpp#L36-L52). Try with something along these lines:

```python
pos.positionMove(joint, angle)

while True:
    if pos.checkMotionDone():
        break
    else:
        yarp.Time.delay(0.5)

print('Done!')
```

jgvictores, 2017-05-24:
I'd also recommend a small delay before the first check:

```python
pos.positionMove(joint, angle)

yarp.Time.delay(0.5)

while True:
    if pos.checkMotionDone():
        break
    else:
        yarp.Time.delay(0.5)

print('Done!')
```

Reimplementing this in a more like a do-while fashion:

```python
pos.positionMove(joint, angle)

while True:
    yarp.Time.delay(0.5)
    if pos.checkMotionDone():
        break

print('Done!')
```

PeterBowman, 2017-05-27:
Excuse me, I was wrong. The value returned by `checkMotionDone` (as well as many other controlboard interface methods) indicates whether the RPC command succeeded or failed. You still have to pass an output parameter to store the actual result of the query. It's easier to see when requesting a position movement, although we are dealing with input parameters here. From inline documentation at `yarp.py`:

```
positionMove(IPositionControl self, DVector data) -> bool
```

That means: pass a `yarp.DVector` containing the desired encoder angles and return a boolean on success or failure. Similarly, at `checkMotionDone` the two available signatures are described as follows:

```
checkMotionDone(IPositionControl self, BVector flag) -> bool
checkMotionDone(IPositionControl self, int i, BVector flag) -> bool
```

We learn that this method accepts a `yarp.BVector` (vector of booleans). In fact, this collection **will always store a single value**, no matter how many joints your robot has. The former retrieves the motion status for all joints, whereas the latter refers only to joint `i`.

Fixed code including previous amendments by @jgvictores:

```python
pos.positionMove(joint, angle)

while True:
    yarp.Time.delay(0.5)
    motionStatus = yarp.BVector(1)
    
    if not pos.checkMotionDone(motionStatus):
        print('Unable to request motion status')
        break
    
    if motionStatus[0]:
        break

print('Done!')
```

Alternatively, the total wait time may be calculated ahead and compared on each iteration with the accumulated duration of the movement:

```python
import math

# ... obtain current encoder angle and reference speed for motor 'joint'
maxTime = math.fabs(currentAngle - targetAngle) / speed
startTime = yarp.Time.now()
pos.positionMove(joint, targetAngle)

while True:
    yarp.Time.delay(0.5)
    if yarp.Time.now() - startTime > maxTime:
        break

print('Done!')
```

PeterBowman, 2017-05-28:
I've deleted my comment from last night (must have suffered from brain overheat, dunno). Let me summarize my conclusions:

* There are three signatures for the Python bindings of `IPositionControl::checkMotionStatus` and its overloads. Looking at the [SWIG interface file](https://github.com/robotology/yarp/blob/3e63e13/bindings/yarp.i#L819-L844):
  * `checkMotionDone(IPositionControl self) -> bool`: returns the motion status for all joints (finished == `True`) and `True` on failure
  * `checkMotionDone(IPositionControl self, BVector flag) -> bool`: returns `True` or `False` on success/failure, and stores a **single boolean value** in a `yarp.BVector` container describing the motion status of all joints
  * `checkMotionDone(IPositionControl self, int i, BVector flag) -> bool`: like above, but this one refers only to joint `i`

* Example usage of the second signature:
  ```python
  pos.positionMove(joint, angle)
  
  while True:
      yarp.Time.delay(0.5)
      motionStatus = yarp.BVector(1)
      
      if not pos.checkMotionDone(motionStatus):
          print('Unable to request motion status')
          break
      
      if motionStatus[0]:
          break
  
  print('Done!')
  ```

* Alternatively, the total wait time may be calculated ahead and compared on each iteration with the accumulated duration of the movement:
  ```python
  import math
  
  # ... obtain current encoder angle and reference speed for motor 'joint'
  maxTime = math.fabs(currentAngle - targetAngle) / speed
  startTime = yarp.Time.now()
  pos.positionMove(joint, targetAngle)
  
  while True:
      yarp.Time.delay(0.5)
      if yarp.Time.now() - startTime > maxTime:
          break
  
  print('Done!')
  ```

serna92, 2017-05-29:
Thank you to you both for resolving my doubts. I recorded a video demostration with the asibot simulation running. I also include the repository where I storage the code.

[youtube video](https://youtu.be/wsg1LLEc5C0)
[Asibot Simulation repository](https://github.com/serna92/AsibotSimulation)

jgvictores, 2017-05-31:
- [youtube video](https://youtu.be/wsg1LLEc5C0) -> Added to playlist: https://www.youtube.com/watch?v=wsg1LLEc5C0&list=PL9C7CA77199E91668
- [Asibot Simulation repository](https://github.com/serna92/AsibotSimulation) -> Starred :-)

Ok to close issue?

serna92, 2017-05-31:
Yes :)

closed 2017-05-31 by jgvictores