roboticslab-uc3m/asibot-main #36: Change robot position without closing a cartesianserver instance.

serna92, 2017-05-23:
Hello,

I did an [Asibot user interface](https://github.com/serna92/AsibotSimulation/blob/replaceCartesianServer/programas/gui.py) to load each of the tasks I implemented. These tasks has the chair and the robot in different positions in order to be in range to grab release the different objects.

Before launch each task, I modify the position of the objects, the robot and the wheelchair [reading and writing the xml file](https://github.com/serna92/AsibotSimulation/blob/replaceCartesianServer/programas/gui.py#L219),

I can delete and add again the objects but in the case of the robot I get an error if cartesianServer is launched without it. It's possible to reload the environment or change the position of the robot once cartesianServer is loaded?.

Thanks.

jgvictores, 2017-05-25:
Do you think you could you add a `world mv (string)` command to [src/libraries/RlPlugins/ravebot/WorldRpcResponder.cpp](https://github.com/roboticslab-uc3m/asibot-main/blob/develop/src/libraries/RlPlugins/ravebot/WorldRpcResponder.cpp) and PR?

serna92, 2017-05-25:
Yeah, it seems the best solution. Is there any difference between moving robot and moving any kinbody object like the weelchair? I saw translation values are stored in [asibot.robot.xml](https://github.com/serna92/AsibotSimulation/blob/replaceCartesianServer/entornoAsibot/asibot.robot.xml).

jgvictores, 2017-05-25:
As far as I know, it should be the same. I expect if we force the kinbody of the ASIBOT link used as base to a different place, the rest of the links should follow. Of course, this is an assumption, we should test it.

The contents of the new `world mv (string)` command in [src/libraries/RlPlugins/ravebot/WorldRpcResponder.cpp](https://github.com/roboticslab-uc3m/asibot-main/blob/develop/src/libraries/RlPlugins/ravebot/WorldRpcResponder.cpp) should be something similar to [this oldie](https://github.com/roboticslab-uc3m/asibot-main/blob/b096cc52a264e0cc18568335a80fb5f0008c6f79/src/libraries/RavePlugins/externobj/ExternObj.cpp#L62-L67).

jgvictores, 2017-05-25:
@serna92 Maybe we can close this thanks to #37 ?

serna92, 2017-05-25:
Yes, I agree.

closed 2017-05-25 by serna92