roboticslab-uc3m/asibot-main #38: Clean asibot-main examples

jgvictores, 2017-11-04:
Clean asibot-main examples

PeterBowman, 2017-11-04:
@jgvictores I forgot to tell you that there is ongoing work (#21, #28) on the `ycm-migration` branch. If you could, please work against it to keep track of the overall progress in migrating/cleaning files. I have just synced it `develop`.

jgvictores, 2017-11-04:
No problem. I'm essentially just going to be deleting certain files from the `example` folder. Would you prefer if I do this directly on `ycm-migration` (if so, I guess on a branch and then merge --no-ff)?

PeterBowman, 2017-11-04:
Oh, I didn't notice that I never actually touched the `example/` directory at all. No problem then if you choose to merge into `develop`, I'll keep up later with your changes on `ycm-migration`.

jgvictores, 2018-02-03:
Done all but TaskLib at 71146f80b9c1bcd1675fb90ddeb7253fa528f6a9

closed 2018-02-03 by jgvictores